jess-moss/koch-v1-1 #6: solidworks file format

Kevin-spark, 2024-08-22:
Thank you for your excellent work. When I tried to open the project's SolidWorks files, I received a version mismatch warning. Could you please let me know which version of SolidWorks you are using? Also, would it be possible to save the files in STEP format? The STP format is more universally compatible for 3D CAD file exchange.

I would greatly appreciate it if you could assist with this. Thank you!

jess-moss, 2024-08-22:
I am using SolidWorks 2024. 

I can definitely save a STEP version but it will take a few days. In the meantime hopefully you can at least use the STL files (which you can also open with SW).

Hope this helps!

haoyangli16, 2024-09-04:
Hi @jess-moss ,

Really a nice and useful work! But could you please save the SolidWorks files in an earlier version? Many engineers, including myself, aren't using the latest version. Alternatively, providing a STEP file would make it even more accessible across different software.

Thanks so much for your consideration and for the great work!

jess-moss, 2024-09-04:
Hello @haoyangli16-- thank you so much! What version of SW do you recommend? I can pretty easily save these files as SW22 or  SW23. Would that be sufficient? In addition, it is on my list to save them as a step file- I just have not gotten around to it yet. Since a few people have asked, I will try and do that by the end of the week. Thank you!

jess-moss, 2024-09-05:
@haoyangli16 @Kevin-spark just added the step files. Hope this helps!

closed 2024-12-05 by jess-moss