jess-moss/koch-v1-1 #13: Cannot update firmware on XL-430 via waveshare servo driver

codenamev, 2024-09-19:
Hey there!  Firstly, thank you for an awesome project ❤️ 

I was able to fully assemble everything and ran a successful teleoperation as part of [the LeRobot calibration step](https://github.com/huggingface/lerobot/blob/main/examples/7_get_started_with_real_robot.md#d-use-kochyaml-and-our-teleoperate-function).  However, when it came time to [run it with cameras enabled](https://github.com/huggingface/lerobot/blob/main/examples/7_get_started_with_real_robot.md#d-use-kochyaml-and-our-teleoperate-function), the arm swung up and the top XL-430 started flashing red.  After that, I verified all the connections and proper voltage, but was unable to recalibrate these.

I thought it could be a firmware issue, and downloaded [the Dynamixel Wizard](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/).  Unfortunately, [there are known issues updating the firmware over USB-C with the Waveshare servo driver](https://forum.robotis.com/t/firmware-transfer-failed-on-new-motor/5819/18).  I ordered their [official U2D2](https://www.robotis.us/u2d2/) (coming today) and will update this to see if that resolves the issue, but wanted to create this in case others have issues (might be worth a mention in the README).

PradeepKadubandi, 2024-11-28:
Just to add my experience, I had the same trouble using Mac / Windows software. But when I tried it on Linux, I was able to use the same board to recover the firmware.

I second the "awesome" comment! Thank you LeRobot team and @jess-moss . This repository is much more detailed and helpful to be able to assemble and use these arms.

jess-moss, 2024-12-05:
Thank you so much!! 

That is an interesting thought. I know the Waveshare boards do commonly have some issues and are definitely not as robust as the official Robotis board so that is a good step. I personally use linux and did not encounter problems updating but will keep this issue open so others are aware.