jess-moss/koch-v1-1 #14: Leader arm base is different on video x github

diegoakel, 2024-10-07:
This base:

[image]

Is different from the one on the hardware/leader folder.

[image]

includingThis, 2024-10-27:
We had no issues assembling the parts nevertheless. However, in the video the servo is placed in a position turned 90 degrees. May be related to different base layout. Drawing in the manual confirms correct servo position parallel to the long side of the base.

diegoakel, 2024-10-27:
Yes, not a real problem to be honest. Maybe just a version difference between video and repo. I guess this issue could be closed.

closed 2024-10-28 by diegoakel